epics-modules/xxx #12: BASH_SOURCE undefined during bash script execution

prjemian, 2018-11-29:
Running `xxx/.../xxx.sh` gives this non-fatal error message before proceeding

```
dirname: missing operand
Try 'dirname --help' for more information.
```

This comes from executing this subcommand:  `dirname ${BASH_SOURCE}`

Insertion of an `echo "|${BASH_SOURCE}|"` line yields `||` indicating that `BASH_SOURCE` is undefined during execution of this script.  Since the `BASH_SOURCE` array is undefined, the `dirname` command is called with no argument resulting in the message above.

prjemian, 2018-11-29:
Use of `BASH_SOURCE` is [risky](https://www.ostricher.com/2014/10/the-right-way-to-get-the-directory-of-a-bash-script/), prone to failures such as this one.

Suggest replacing use of `BASH_SOURCE` in synApps with a shell variable that is known to exist, such as `SNAME` in `xxx.sh`.  Here are the instances of `BASH_SOURCE` I found in a fresh synApps clone:

```
mintadmin@mint-vm ~/.../synApps/support $ !grep
grep -R BASH_SOURCE *
areaDetector-R3-3-1/ADSupport/supportApp/GraphicsMagickSrc/png/configure:      BASH_ARGV | BASH_SOURCE) eval $ac_var= ;; #(
areaDetector-R3-3-1/ADSupport/supportApp/GraphicsMagickSrc/png/configure:      BASH_ARGV | BASH_SOURCE) eval $ac_var= ;; #(
areaDetector-R3-3-1/ADSupport/supportApp/GraphicsMagickSrc/lcms/configure:      BASH_ARGV | BASH_SOURCE) eval $ac_var= ;; #(
areaDetector-R3-3-1/ADSupport/supportApp/GraphicsMagickSrc/lcms/configure:      BASH_ARGV | BASH_SOURCE) eval $ac_var= ;; #(
areaDetector-R3-3-1/ADSupport/supportApp/GraphicsMagickSrc/webp/configure:      BASH_ARGV | BASH_SOURCE) eval $ac_var= ;; #(
areaDetector-R3-3-1/ADSupport/supportApp/GraphicsMagickSrc/webp/configure:      BASH_ARGV | BASH_SOURCE) eval $ac_var= ;; #(
xxx-R6-0/start_MEDM_xxx:export EPICS_APP=`dirname $BASH_SOURCE`
xxx-R6-0/start_caQtDM_xxx:export EPICS_APP=`dirname $BASH_SOURCE`
```

I'll fix the ones in xxx and accept the ones in GraphicsMagickSrc without change.  The `make` did not fail for that code anyway.

kmpeters, 2018-11-29:
@prjemian, I see that this issue, as well as #13 and #17 are problems on Linux Mint. Do these problems also exist on RHEL7?

prjemian, 2018-11-29:
Works on RHEL7

Here's my test code (`kevin.sh`):

```
#!/bin/sh

echo "BASH_SOURCE = ${BASH_SOURCE}"
PATH=`dirname ${BASH_SOURCE}`
echo "PATH = ${PATH}"
```

Now running it:

```
jemian@otz ~/sandbox $ ./kevin.sh 
BASH_SOURCE = ./kevin.sh
PATH = .
jemian@otz ~/sandbox $ pwd
/home/oxygen/JEMIAN/sandbox
jemian@otz ~/sandbox $ pushd /tmp
jemian@otz /tmp $ /home/oxygen/JEMIAN/sandbox/kevin.sh 
BASH_SOURCE = /home/oxygen/JEMIAN/sandbox/kevin.sh
PATH = /home/oxygen/JEMIAN/sandbox
```

prjemian, 2018-11-29:
Another way to resolve would be to use `$0` as a fallback case, such as:

```
#!/bin/sh

echo "xBASH_SOURCE = ${xBASH_SOURCE}"
PATH=`dirname ${xBASH_SOURCE:-$0}`
echo "PATH = ${PATH}"
```

Looks like this:

```
jemian@otz /tmp $ /home/oxygen/JEMIAN/sandbox/kevin.sh 
xBASH_SOURCE = 
PATH = /home/oxygen/JEMIAN/sandbox
```

prjemian, 2018-11-29:
See 3234d66

PR #14 (2018-12-14): BASH scripting changes
fixes #12
commit b4264682 (2018-12-14): fixes #12
closed 2018-12-14 by kmpeters