epics-modules/xxx #18: motorSim LLM same as HLM

prjemian, 2018-12-04:
With synApps 6.0 and base 7.0.2, enabling the motorSim soft motors

(adding `motors.cmd` to st.cmd.Linux and enabling this `iocshLoad("$(MOTOR)/iocsh/motorSim.iocsh", "INSTANCE=motorSim, HOME_POS=0, NUM_AXES=16, SUB=substitutions/motorSim.substitutions")` and removing the `substitutions/motor.substitutions`), both LLM and HLM come up as identical.  Since they are non-zero *and* the motor is at zero, the soft motor cannot move until the lower limit is changed.  All 16 soft motors have this problem.

prjemian, 2018-12-04:
This can be *patched* by adding `LOW_LIM=-32000` to the `motors.cmd` file on the line with `iocshLoad("$(MOTOR)/iocsh/motorSim.iocsh", ` ...

The real problem is in the source of the `motorSim.iocsh`, on this line:

    motorSimCreate($(CONTROLLER=0), 0, $(LOW_LIM=-32000), $(HIGH_LIM=32000), $(HOME_POS=0), 1, $(NUM_AXES=1))

The default value for `LOW_LIM`, when the symbol is not declared, becomes positive.

prjemian, 2018-12-04:
I'll add this patch for now but we really should get the default parameters to be set properly.

closed 2018-12-04 by prjemian

kmpeters, 2018-12-11:
This issue might be resolved by this commit https://github.com/epics-modules/motor/commit/df11f3e153e2aa6c6cb605cbc1392e0a9e663d93

prjemian, 2018-12-11:
agreed

commit 61ff2835 (2018-12-04): fixes #18