epics-modules/xxx #24: BUG: bash syntax incorrect in starter script

prjemian, 2019-09-02:
in IOC built from xxx 6.1:

```
(base) mintadmin@mint-vm:.../iocvm7/softioc$ ./vm7.sh stop
./vm7.sh: 110: ./vm7.sh: [[: not found
./vm7.sh: 115: ./vm7.sh: [[: not found
./vm7.sh: 110: ./vm7.sh: [[: not found
./vm7.sh: 115: ./vm7.sh: [[: not found```

prjemian, 2019-09-02:
See this source: https://github.com/epics-modules/xxx/blob/e798cbdecc3d394436dd2f9b504d27f59f6b6d32/iocBoot/iocxxx/softioc/xxx.sh#L110-L118

Fix is to use single `[`, not double `[[`:

```
                            if [ ${s_pid} -eq ${P_PID} ] ; then
                                SCREEN_PID=${s_pid}
                                break
                            fi
                    
                            if [ ${s_pid} -eq ${GP_PID} ] ; then
                                SCREEN_PID=${s_pid}
                                break
                            fi
```

kmpeters, 2019-09-03:
Do the variables need to be in quotes when single square braces are used?

prjemian, 2019-09-03:
Not required but great advice

PR #25 (2019-09-30): use single square braces here
fixes #24
closed 2019-09-30 by keenanlang