epics-modules/xxx #26: can't find motorSim.iocsh

prjemian, 2019-09-02:
Tried using the simulated motor support and ran into this:

```
iocshLoad("/usr/local/epics/synApps/support/motor-R7-1/iocsh/motorSim.iocsh", "INSTANCE=motorSim, HOME_POS=0, NUM_AXES=16, SUB=substitutions/motorSim.substitutions")
Can't open /usr/local/epics/synApps/support/motor-R7-1/iocsh/motorSim.iocsh: No such file or directory
```

kmpeters, 2019-09-03:
This is an unresolved problem in motor, after the split; ``.iocsh`` files in driver submodules do not get installed to a top-level motor directory.  @anjohnson suggested that a module's top-level iocsh directory should contain installed iocsh files, which is the first step to resolving this issue:

https://github.com/epics-modules/motor/issues/55#issuecomment-474004365

It isn't obvious to me where ``.iocsh`` files should reside.  Having a discussion with @keenanlang about it is still on the to-do list.

keenanlang, 2019-09-03:
> belongs in xxx

Does it? 

The whole point of moving them out of xxx is so that ioc shell support/examples can reside with the module support, thus allowing it to be updated side-by-side with changes to the underlying support. If a fix for an issue happens, the iocsh file can be updated and then xxx startup files wouldn't have to change at all to get the new behavior. Rather than how we had it previously where the examples in the cmd files were frequently out-of-date from how the module currently worked. 

Is what is going on in motorSim.iocsh significantly outside the realm of the motor module that it would require movement into xxx? It seems the issue is just that we need to figure out makefile rules to copy all the submodule iocsh files into Motor's top-level iocsh directory.

kmpeters, 2019-09-03:
> > belongs in xxx
> Does it?

As long as xxx has .iocsh files in the examples directory, we should probably eliminate errors in them:

https://github.com/epics-modules/xxx/blob/master/iocBoot/iocxxx/examples/motors.iocsh

> It seems the issue is just that we need to figure out makefile rules to copy all the submodule iocsh files into Motor's top-level iocsh directory.

I would vote for Andrew's suggestion to put the .iocsh files somewhere else and install them to the top-level iocsh directory.  It would be confusing if motor's top-level iocsh dir contained both resident and installed .iocsh files.

Submodules could install to the parent module's top-level dir or to their own top-level dir when built standalone, just like the databases and req files do now.

I don't, however, have any good suggestions for where .iocsh files should reside before they are installed.

keenanlang, 2019-09-03:
>As long as xxx has .iocsh files in the examples directory, we should probably eliminate errors in them:

Absolutely. I was just addressing the idea that motorSim.iocsh belongs in xxx. 

>I don't, however, have any good suggestions for where .iocsh files should reside before they are installed.

<module/submodule>/iocBoot/<whatever> ?

anjohnson, 2019-09-03:
You can’t install .iocsh files from an iocBoot directory, they would have to go in to a src or Db directory, then you can use the same technique Keenan was using for .req files. Sorry it’s late over here and I don’t have my laptop out right now to give more details.

PR #27 (2019-09-30): catchup with changes in motor module
fixes #26
closed 2019-09-30 by keenanlang