epics-modules/xxx #30: allow a custom IOC prefix

prjemian, 2019-09-30:
It *could be* easy to specify an alternate IOC prefix, yet default to `xxx:` (even easier once #29 is merged) if we changed on line in `common.iocsh`: https://github.com/epics-modules/xxx/blob/1bae0bb1e60d5011aa1c1245ecececf8ffecec57/iocBoot/iocxxx/common.iocsh#L5

with this replacement (that works on Linux):

    epicsEnvSet("PREFIX", "$(PREFIX:-xxx:)")

For details on this technique, see: https://www.cyberciti.biz/tips/bash-shell-parameter-substitution-2.html

Will this work on ALL operating systems?

keenanlang, 2019-09-30:
>Will this work on ALL operating systems?

I find it interesting that it's even working on Linux. Bash isn't what's processing the commands, it's the iocsh and the iocsh uses the macLib library in base. As far as I can tell, that line should cause an error.

For something that will work with anything, just set a default for the macro substitution

    $(PREFIX=xxx:)

prjemian, 2019-09-30:
When I said it works on Linux, I mean from the linux command line, not the IOC shell.  It does not work there and consensus from yesterday's tech-talk email says it is not possible.  Maybe this can be done with some `luash` function?

For a custom IOC prefix, a default will not work.  Want to start the IOC using a user-defined IOC prefix with a default value of `xxx:`.  Most synApps databases follow the practice to use a ":" as a delimiter and the prefix should include the trailing colon.  The sole exception to this practice is:

    dbLoadRecords("$(DEVIOCSTATS)/db/iocAdminSoft.db","IOC=xxx")

where `iocAdminSoft.db` provides the colon **after** the $(IOC), such as `$(IOC):UPTIME`.  When the previous line is changed to 

    dbLoadRecords("$(DEVIOCSTATS)/db/iocAdminSoft.db","IOC=$PREFIX")

this results in PVs with two adjacent colons, such as `xxx::UPTIME`

Short of getting the maintainers of $(DEVIOCSTATS) to change their database(s), the fix is to supply an edited version of $PREFIX with the trailing colon removed.

anjohnson, 2019-09-30:
Please read the issue report at epics-modules/iocStats#31 and unmerged PR at epics-modules/iocStats#32 that are vaguely related — ESS want to be able to use a separator other than `:`. I'm not too keen on the rather bulky `$(SEP=:)` variable, but there isn't too much alternative, `$(S=:)` would be somewhat shorter but less descriptive.

prjemian, 2019-10-05:
An interim workaround for the double colon issue posed by:

    dbLoadRecords("$(DEVIOCSTATS)/db/iocAdminSoft.db","IOC=xxx")

is to create a database of PV aliases associating together the single-colon aliases with their existing double-colon counterparts.  That is possible now and would improve the current situation, allowing view of the `ioc_stats_soft.*` screens.

prjemian, 2019-10-05:
The list is

```
xxx::GTIM_ERR_CNT
xxx::ST_SCRIPT
xxx::EPICS_VERSION
xxx::APP_DIR
xxx::ACCESS
xxx::CA_CLNT_CNT
xxx::CA_CONN_CNT
xxx::RECORD_CNT
xxx::FD_MAX
xxx::FD_CNT
xxx::SYS_CPU_LOAD
xxx::IOC_CPU_LOAD
xxx::LOAD
xxx::CPU_CNT
xxx::SUSP_TASK_CNT
xxx::MEM_USED
xxx::MEM_FREE
xxx::MEM_MAX
xxx::PROCESS_ID
xxx::PARENT_ID
xxx::SCANONCE_Q_SIZE
xxx::CB_Q_SIZE
xxx::SCANONCE_Q_HIGH
xxx::SCANONCE_Q_USED
xxx::SCANONCE_Q_OVERRUNS
xxx::CBLOW_Q_HIGH
xxx::CBLOW_Q_USED
xxx::CBLOW_Q_OVERRUNS
xxx::CBMEDIUM_Q_HIGH
xxx::CBMEDIUM_Q_USED
xxx::CBMEDIUM_Q_OVERRUNS
xxx::CBHIGH_Q_HIGH
xxx::CBHIGH_Q_USED
xxx::CBHIGH_Q_OVERRUNS
xxx::GTIM_TIME
xxx::CA_UPD_TIME
xxx::FD_UPD_TIME
xxx::LOAD_UPD_TIME
xxx::MEM_UPD_TIME
xxx::FD_FREE
xxx::SCANONCE_Q_HIGHPER
xxx::SCANONCE_Q_USEDPER
xxx::CBLOW_Q_HIGHPER
xxx::CBLOW_Q_USEDPER
xxx::CBMEDIUM_Q_HIGHPER
xxx::CBMEDIUM_Q_USEDPER
xxx::CBHIGH_Q_HIGHPER
xxx::CBHIGH_Q_USEDPER
xxx::STARTTOD
xxx::TOD
xxx::ST_SCRIPT1
xxx::ST_SCRIPT2
xxx::KERNEL_VERS
xxx::EPICS_VERS
xxx::HOSTNAME
xxx::APP_DIR1
xxx::APP_DIR2
xxx::UPTIME
xxx::ENGINEER
xxx::LOCATION
xxx::GTIM_CUR_SRC
xxx::GTIM_EVT_SRC
xxx::GTIM_HI_SRC
xxx::CA_ADDR_LIST
xxx::CA_CONN_TIME
xxx::CA_AUTO_ADDR
xxx::CA_RPTR_PORT
xxx::CA_SRVR_PORT
xxx::CA_MAX_ARRAY
xxx::CA_SRCH_TIME
xxx::CA_BEAC_TIME
xxx::TIMEZONE
xxx::TS_NTP_INET
xxx::IOC_LOG_PORT
xxx::IOC_LOG_INET
xxx::GTIM_RESET
xxx::READACF
xxx::SYSRESET
xxx::SysReset
xxx::HEARTBEAT
xxx::START_CNT
```

prjemian, 2019-10-07:
The relevant section of the EPICS documentation is https://epics.anl.gov/base/R3-16/2-docs/AppDevGuide/DatabaseDefinition.html#x7-2090006.2 -- place lines like this in a database `/.db` file and load with `dbLoadRecords(...)`

    alias(record_name,alias_name)

prjemian, 2019-10-07:
```
alias($(P):GTIM_ERR_CNT,   $(P)GTIM_ERR_CNT)
alias($(P):ST_SCRIPT,      $(P)ST_SCRIPT)
alias($(P):EPICS_VERSION,  $(P)EPICS_VERSION)
```

...

keenanlang, 2019-10-08:
Fixed with #31

closed 2019-10-08 by keenanlang