epics-modules/xxx #32: associate screen files by application, not extension

prjemian, 2019-10-23:
As the EPICS community grows, the number of different GUI display managers increases.  To date, we have organized screens for each display manager inside the `xxxApp/op` directory in subdirectories labeled by the screen file's extension.

Now, we must consider a new display manager that uses the same screen extension as an existing display manager.  Since the files are different internally, yet must have the same name, they cannot be placed in the same directory.

| extension | subdir now | new subdir | display manager |
| ---- | ---- | ---- | ---- |
| `.adl` | adl | medm | [MEDM](https://epics.anl.gov/extensions/medm/index.php) |
| `.edl` | edl | edm | [EDM](https://epics.anl.gov/tech-talk/2019/msg00399.php) |
| `.opi` | opi | boy | [CSS BOY](https://github.com/ControlSystemStudio/cs-studio/wiki/BOY) |
| `.ui` | ui | caqtdm | [caQtDM](https://github.com/caqtdm/caqtdm) |
| `.ui` |  (*) | pydm | [PyDM](https://github.com/slaclab/pydm) |

*(\*) PyDM screen files not yet provided. (An autoconverter from MEDM's `.adl` format is in the works at this time.)*

This new practice of directory names is already in use by the [asyn](https://github.com/epics-modules/asyn/tree/master/opi) module.  We choose at this time not to relocate the screen files to a top-level `opi` directory, as *asyn* has done, since there may still be value to leave them in the associated `___App/op` directory.

prjemian, 2019-10-23:
Once this xxx module is converted, each of the modules referenced in `setup_epics_common` should be edited similarly, with reference to this issue.  Once a module is converted, the `setup_epics_common` script here should be updated with the new subdirectory names.

prjemian, 2019-10-23:
@keenanlang What other code in this module should be modified for this change?

MarkRivers, 2019-10-23:
I understand the conflict with PyDM. However I think that changing the existing names is a bad idea. Many sites will have display paths (e.g. EPICS_DISPLAY_PATH, etc.) using the existing names. They will also be using the existing names in scripts to copy files to a central location. I think we should just use a new name for PyDM.

prjemian, 2019-10-23:
That *would* be less work.  In fact, no work at all until PyDM files (via an autoconverter) are ready to be contributed.

prjemian, 2019-10-23:
I'll push the other edits I have to this branch, just because they are ready.  Looks like we'll delete this branch in the end.

prjemian, 2019-10-23:
The only edit worth keeping, then, is c5e55b7, which changes CRLF to LF in one file (from #31).

closed 2019-10-23 by prjemian