epics-modules/xxx #40: xxx R6-1 does not compile with epics 7.0.4 in windows x64

LeeYangLBLBCS, 2020-07-15:
**Describe the issue**

**Steps to reproduce**
1. compile in windows x64

**Expected behaivour**
no compile error
**Actual behaviour**
compiler error:
gmake -C ./xxxApp install
gmake[1]: Entering directory 'C:/epics/modules/synApps_6_1_epics7/support/xxx-R6-1/xxxApp'
gmake -C ./src install
gmake[2]: Entering directory 'C:/epics/modules/synApps_6_1_epics7/support/xxx-R6-1/xxxApp/src'
gmake -C O.windows-x64 -f ../Makefile TOP=../../.. \
    T_A=windows-x64 install
gmake[3]: Entering directory 'C:/epics/modules/synApps_6_1_epics7/support/xxx-R6-1/xxxApp/src/O.windows-x64'
gmake[3]: *** No rule to make target '../../../lib/windows-x64/Parker.lib', needed by 'xxx.exe'.  Stop.
gmake[3]: Leaving directory 'C:/epics/modules/synApps_6_1_epics7/support/xxx-R6-1/xxxApp/src/O.windows-x64'
gmake[2]: *** [C:\epics\base-7.0.4/configure/RULES_ARCHS:58: install.windows-x64] Error 2
gmake[2]: Leaving directory 'C:/epics/modules/synApps_6_1_epics7/support/xxx-R6-1/xxxApp/src'
gmake[1]: *** [C:\epics\base-7.0.4/configure/RULES_DIRS:85: src.install] Error 2
gmake[1]: Leaving directory 'C:/epics/modules/synApps_6_1_epics7/support/xxx-R6-1/xxxApp'
gmake: *** [C:\epics\base-7.0.4/configure/RULES_DIRS:85: xxxApp.install] Error 2

**Build Environment**
Architecture: windows x64
Epics Base Version: 7.0.4
Dependent Module Versions: xxxR6-1, xxxR6.0.1

MarkRivers, 2020-07-15:
That error is probably because you did not build the Parker support in the motor module. What version of the motor module are you using and which drivers did you build?

LeeYangLBLBCS, 2020-07-15:
you are right. I didn't realize the github tags contains empty directories
for all motor modules.
I should've checked first.
I've been using tagged versions to download various other epics modules
without a problem.
I guess the motors are created differently somehow.
[image: image.png]

MarkRivers, 2020-07-15:
You need to clone epics-modules/motor with the —recursive flag to get the driver repositories from the epics-motors project.

LeeYangLBLBCS, 2020-07-15:
Did that. Everything worked.
Thanks.

closed 2020-07-21 by LeeYangLBLBCS