epics-modules/xxx #41: build error with xxx-R6-1 with EPICS base 7.0.4 on Ubuntu 18.0.4 - missing libmw100.a

LeeYangLBLBCS, 2020-07-21:
I am trying to compile the same synapps package copied from my Windows x64 build on Ubuntu 18.0.4, linux-x86_64.
Most of modules built successfully except with xxx-R6-1, where it complains about a missing library "libmw100.a".
I can't figure out what it is, and what module is causing the build to want to link to it. Apparently it's a linux specific issue since the same pacage built already with Windows x64.
**My CONFIG_SITE settings**:
LINUX_NET_INSTALLED=NO
LINUX_USB_INSTALLED=NO
**My RELEASE file:**
ALIVE=$(SUPPORT)/alive-R1-1-1
ASYN=$(SUPPORT)/asyn-R4-38
AUTOSAVE=$(SUPPORT)/autosave-R5-10
BUSY=$(SUPPORT)/busy-R1-7-2
CALC=$(SUPPORT)/calc-R3-7-4
CAMAC=$(SUPPORT)/camac-R2-7-1
CAPUTRECORDER=$(SUPPORT)/caputRecorder-R1-7-3
DAC128V=$(SUPPORT)/dac128V-R2-9
DELAYGEN=$(SUPPORT)/delaygen-R1-2-1
DXP=$(SUPPORT)/dxp-R6-0
DXPSITORO=$(SUPPORT)/dxpSITORO-R1-2
DEVIOCSTATS=$(SUPPORT)/iocStats-3-1-16
IP=$(SUPPORT)/ip-R2-20-1
IPAC=$(SUPPORT)/ipac-2-15
IP330=$(SUPPORT)/ip330-R2-9
IPUNIDIG=$(SUPPORT)/ipUnidig-R2-11
LOVE=$(SUPPORT)/love-R3-2-7
LUA=$(SUPPORT)/lua-R2-1
MCA=$(SUPPORT)/mca-R7-8
MEASCOMP=$(SUPPORT)/measComp-R2-3
MODBUS=$(SUPPORT)/modbus-R3-0
MOTOR=$(SUPPORT)/motor-R7-2-1
OPTICS=$(SUPPORT)/optics-R2-13-4
QUADEM=$(SUPPORT)/quadEM-R9-2-1
SOFTGLUE=$(SUPPORT)/softGlue-R2-8-2
SOFTGLUEZYNQ=$(SUPPORT)/softGlueZynq-R2-0-2
SSCAN=$(SUPPORT)/sscan-R2-11-3
STD=$(SUPPORT)/std-R3-6-1
STREAM=$(SUPPORT)/StreamDevice-2-8-13
VAC=$(SUPPORT)/vac-R1-9
VME=$(SUPPORT)/vme-R2-9-2
YOKOGAWA_DAS=$(SUPPORT)/Yokogawa_DAS-R2-0-1
XXX=$(SUPPORT)/xxx-R6-1
AREA_DETECTOR=$(SUPPORT)/areaDetector-R3-7
ADCORE=$(AREA_DETECTOR)/ADCore
ADSUPPORT=$(AREA_DETECTOR)/ADSupport
ADSIMDETECTOR=$(AREA_DETECTOR)/ADSimDetector
SNCSEQ=$(SUPPORT)/seq-2-2-6
ALLEN_BRADLEY=$(SUPPORT)/allenBradley-2-3
=========================================
**Build error output**
/usr/bin/gcc  -D_GNU_SOURCE -D_DEFAULT_SOURCE           -D_X86_64_  -DUNIX  -Dlinux      -g   -Wall      -mtune=generic      -m64  -I. -I../O.Common -I. -I. -I.. -I../../../include/compiler/gcc -I../../../include/os/Linux -I../../../include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/include -I/home/lyang/epics/base-7.0.4/include/compiler/gcc -I/home/lyang/epics/base-7.0.4/include/os/Linux -I/home/lyang/epics/base-7.0.4/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/alive-R1-1-1/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/asyn-R4-38/include  -I/home/lyang/epics/modules/synApps_6_1_epics7/support/autosave-R5-10/include/os/Linux -I/home/lyang/epics/modules/synApps_6_1_epics7/support/autosave-R5-10/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/busy-R1-7-2/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/calc-R3-7-4/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/camac-R2-7-1/include            -I/home/lyang/epics/modules/synApps_6_1_epics7/support/dxp-R6-0/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/dxpSITORO-R1-2/include  -I/home/lyang/epics/modules/synApps_6_1_epics7/support/iocStats-3-1-16/include/os/Linux -I/home/lyang/epics/modules/synApps_6_1_epics7/support/iocStats-3-1-16/include      -I/home/lyang/epics/modules/synApps_6_1_epics7/support/ipac-2-15/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/ip330-R2-9/include         -I/home/lyang/epics/modules/synApps_6_1_epics7/support/lua-R2-1/include  -I/home/lyang/epics/modules/synApps_6_1_epics7/support/mca-R7-8/include/os/Linux -I/home/lyang/epics/modules/synApps_6_1_epics7/support/mca-R7-8/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/measComp-R2-3/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/modbus-R3-0/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/motor-R7-2-1/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/optics-R2-13-4/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/quadEM-R9-2-1/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/softGlue-R2-8-2/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/softGlueZynq-R2-0-2/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/sscan-R2-11-3/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/std-R3-6-1/include      -I/home/lyang/epics/modules/synApps_6_1_epics7/support/vac-R1-9/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/vme-R2-9-2/include            -I/home/lyang/epics/modules/synApps_6_1_epics7/support/areaDetector-R3-7/ADCore/include  -I/home/lyang/epics/modules/synApps_6_1_epics7/support/areaDetector-R3-7/ADSupport/include/os/Linux -I/home/lyang/epics/modules/synApps_6_1_epics7/support/areaDetector-R3-7/ADSupport/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/areaDetector-R3-7/ADSimDetector/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/seq-2-2-6/include   -I/home/lyang/epics/modules/synApps_6_1_epics7/support/allenBradley-2-3/include        -c ../xxxMain.c
**make[4]: *** No rule to make target '../../../lib/linux-x86_64-debug/libmw100.a', needed by 'xxx'.  Stop.**
make[4]: Leaving directory '/home/lyang/epics/modules/synApps_6_1_epics7/support/xxx-R6-1/xxxApp/src/O.linux-x86_64-debug'
/home/lyang/epics/base-7.0.4/configure/RULES_ARCHS:58: recipe for target 'install.linux-x86_64-debug' failed
make[3]: *** [install.linux-x86_64-debug] Error 2
make[3]: Leaving directory '/home/lyang/epics/modules/synApps_6_1_epics7/support/xxx-R6-1/xxxApp/src'
/home/lyang/epics/base-7.0.4/configure/RULES_DIRS:85: recipe for target 'src.install' failed
make[2]: *** [src.install] Error 2
make[2]: Leaving directory '/home/lyang/epics/modules/synApps_6_1_epics7/support/xxx-R6-1/xxxApp'
/home/lyang/epics/base-7.0.4/configure/RULES_DIRS:85: recipe for target 'xxxApp.install' failed
make[1]: *** [xxxApp.install] Error 2
make[1]: Leaving directory '/home/lyang/epics/modules/synApps_6_1_epics7/support/xxx-R6-1'
/home/lyang/epics/base-7.0.4/configure/RULES_DIRS:85: recipe for target '/home/lyang/epics/modules/synApps_6_1_epics7/support/xxx-R6-1.install' failed
make: *** [/home/lyang/epics/modules/synApps_6_1_epics7/support/xxx-R6-1.install] Error 2

prjemian, 2020-07-21:
That's from the yokogawa support.  Do you have a Yokogawa mw100?  If not, remove from both Windows and Linux builds.  Comment out in `/home/lyang/epics/modules/synApps_6_1_epics7/support/configure/RELEASE`

You might examine each of the packages to be built to see if they will not be used on your OS.  For example, the VME package is obvious to not build for Windows or Linux.  For example, these are the ones _I_ do not build for Linux.  Most because I do not have that hardware, the VME because it is not Windows or Linux:

```
# ALLEN_BRADLEY=$(SUPPORT)/allenBradley-2-3
# DXP=$(SUPPORT)/dxp-R6-0
# DXPSITORO=$(SUPPORT)/dxpSITORO-R1-2
# LOVE=$(SUPPORT)/love-R3-2-7
# SOFTGLUE=$(SUPPORT)/softGlue-R2-8-2
# SOFTGLUEZYNQ=$(SUPPORT)/softGlueZynq-R2-0-2
# VME=$(SUPPORT)/vme-R2-9-2
# YOKOGAWA_DAS=$(SUPPORT)/Yokogawa_DAS-R2-0-1
```

LeeYangLBLBCS, 2020-07-21:
thanks.
That's exactly the problem.
I removed those none applicable items in the RELEASE file, the build
succeeded!

prjemian, 2020-07-21:
close this issue once you are satisfied

closed 2020-07-21 by LeeYangLBLBCS