epics-modules/xxx #42: ISSUE NAME HERE

beldish, 2020-08-19:
**Describe the issue**
make of synApps blocks if XXX is in Makefile modules list.

**Steps to reproduce**
1. Step one
Download  synApps_6_1.tar.gz  from https://www.aps.anl.gov/BCDA/synApps/Where-to-find-it

2. Step two
gunzip untar

3. Step three
Run make.

**Expected behaivour**
Start build
**Actual behaviour**
make blocks
**Build Environment**
Architecture:
Scientific Linux 7.8 and Centos 8. Tried on both
Epics Base Version:
git clone --recursive https://github.com/epics-base/epics-base.git 
on 19/08/2020 Branch 7.0
Dependent Module Versions:

beldish, 2020-08-19:
If I comment out XXX - the build starts.

MarkRivers, 2020-08-19:
Please post any error messages.

beldish, 2020-08-19:
There is none, it just blocks.
I traced the makefile execution, it iterates on MODULE_defined for every module and stops on XXX.

This is an output if I add info to Makefile
define  MODULE_defined
  ifdef $(1)
$(info $(1))
  SUPPORT_DIRS  += $($(1))

Output:
[mbeldis@sl7 support]$ make release
ALLEN_BRADLEY
ALIVE
AREA_DETECTOR
ASYN
AUTOSAVE
BUSY
CALC
CAMAC
CAPUTRECORDER
DAC128V
DELAYGEN
DEVIOCSTATS
DXP
DXPSITORO
ETHERIP
IPAC
IP
IP330
IPUNIDIG
LOVE
LUA
MCA
MEASCOMP
MODBUS
MOTOR
OPTICS
QUADEM
SNCSEQ
SOFTGLUE
SOFTGLUEZYNQ
SSCAN
STD
STREAM
VAC
VME
XXX

and it blocks here.

beldish, 2020-08-19:
Looks like the script .../EPICS/epics-base/lib/perl/EPICS/Release.pm
in  sub expandMacros  gets in infinite loop in while loop on line 98.
And this is a case for XXX module only.

MarkRivers, 2020-08-19:
I have seen infinite loops when building many different modules if I rename a directory (e.g. asyn4-30 to asyn4-31) without doing “make clean” or at least “make depsclean”.

beldish, 2020-08-19:
I have not done any renaming. And always do: make clean inunstall; make release and make install.

in .../synApps_6_1/support/xxx-R6-1/configure/RELEASE 
there is a recursive assignment to SUPPORT 

SUPPORT=$(SUPPORT)

This makes the above loop to run in an infinite loop.
If I leave only SUPPORT=,   'make release' assigns a correct value to it.
BTW if I define EPICS_BASE= (which is missing) it will be also correctly defined, but not sure if it is needed or not, as never got to point to compile it. Have other compilation errors in  areaDetector-R3-7/ADCore/ADApp. But this is a different issue.

This issue can be closed as far as I am concerned. Thank you.

MarkRivers, 2020-08-19:
> BTW if I define EPICS_BASE= (which is missing) it will be also correctly defined, but not sure if it is needed or not, as never got to point to compile it.

You must compile EPICS base before you build synApps. Did you do that?

beldish, 2020-08-19:
Yes, I did, I had EPICS base compiled first thing before downloading synApps.

I was talking about missing EPICS_BASE declaration in XXX configure/RELEASE.

beldish, 2020-08-20:
On the second try, I downloaded assemble_synApps.sh from  https://github.com/EPICS-synApps/support.
Configured it and everything worked like a charm.

The version from  https://www.aps.anl.gov/BCDA/synApps/Where-to-find-it also can be compiled if one fixes xxx configure/RELEASE SUPPORT recursion and adds
#include <asynDriver.h>
and/or
#include <epicsExport.h>
to  NDArrayRing.cpp in .../EPICS/support/synApps_6_1/support/areaDetector-R3-7/ADCore/ADApp/pluginSrc

These versions obviously differ as the github version required the installation of additional packages libnet-devel, libcap-devel and libpcap-devel. 
Closing this issue and thank you for your time.

closed 2020-08-20 by beldish