epics-modules/xxx #66: `xxBusyRecord2x.adl` needs translation to CS-Studio `.opi` format.

prjemian, 2023-11-03:
One last holdout: `xxBusyRecord2x.adl` needs translation to CS-Studio `.opi` format.  Cannot assemble a working tool chain today to accomplish that.  Opening in new issue so this work can be merged, noting that file `xxxApp/op/opi/autoconvert/xxBusyRecord2x.opi` is missing and needs translation from the source `.adl` file.

_Originally posted by @prjemian in https://github.com/epics-modules/xxx/issues/65#issuecomment-1793023461_

MarkRivers, 2023-11-03:
I just did the following:
- cd support/xxx
- git pull
- cd xxxApp/op
- make -sj
- git commit -m "New autoconvert file" .
- git push

So the latest autoconvert files are on Github.  Note that I have disabled caQtDM conversions because I don't have a recipe to build it on CentOS 9.

prjemian, 2023-11-03:
Thanks @MarkRivers !

closed 2023-11-03 by prjemian