epics-modules/xxx #71: Perl startup script in synApps 6.3 does not have executable permissions by default

jkirchman, 2024-02-21:
The change in synApps 6.3 from using a shell script to a perl script (xxx.pl) for softioc start up should have set the executable permisions for the new perl script executable. As it is now after installing xxx, the user must manually chmod +x the script file to begin using it (without prefacing it with "perl").

keenanlang, 2024-02-22:
> klang@kobold:~/Documents/Scratch$ git clone git@github.com:epics-modules/xxx.git
> klang@kobold:~/Documents/Scratch$ ls -l xxx/iocBoot/iocxxx/softioc/
> total 20K
> drwxrwxr-x 2 klang aesbc 4.0K Feb 22 13:12 commands
> drwxrwxr-x 4 klang aesbc 4.0K Feb 22 13:12 logs
> -rwxrwxr-x 1 klang aesbc  263 Feb 22 13:12 in-screen.sh
> -rwxrwxr-x 1 klang aesbc  337 Feb 22 13:12 run
> -rwxrwxr-x 1 klang aesbc 2.8K Feb 22 13:12 xxx.pl

xxx.pl seems to already be executable. Is this after running changePrefix? I think the issue is with changes I had to make for that to be cross-compatible. I've just edited that to retain file permissions.

https://github.com/EPICS-synApps/support/commit/5774e8accd88c687d569e1c301014c0780a0620c

jkirchman, 2024-02-27:
Yes, I think you might be correct. A new IOC built today has the x permissions set.

closed 2024-02-27 by jkirchman